eclipse-threadx/usbx #91: USBX send same HID descriptor data for different interface

HaoboGu, 2023-04-03:
Hello, I'm trying to use 2 interfaces in a HID class. My descriptor is like: 
[image]

There are two hid reports, using 2 interfaces. I enabled 2 interfaces by calling `ux_device_stack_class_register` twice, with different `ux_device_class_hid_parameter_report_address` and `interface_number`:
```c
  ux_device_stack_class_register(_ux_system_slave_class_hid_name, ux_device_class_hid_entry,
     0, 0, &parameter1);

  ux_device_stack_class_register(_ux_system_slave_class_hid_name, ux_device_class_hid_entry,
     0, 1, &parameter2);
```

 USBX sends correct configuration descriptor, but wrong HID descriptor data. 

The HID descriptor data for the first interface is correct, but for the second interface, USBX sends the same HID data with the length of second HID report. 

The correct HID descriptor data for the first interface: 
[image]

The wrong HID descriptor data for the second interface(this interface has two endpoints, this data appears twice):
[image]

Except HID descriptor for the second interface(Ignore vid and pid):
[image]

xiaocq2001, 2023-04-06:
I'm not able to reproduce the issue in latest code.
Note in 6.1.8 release there is a fix about such issue: https://github.com/azure-rtos/usbx/releases/tag/v6.1.8_rel (HID descriptor search fix), file modification ([common/usbx_device_classes/src/ux_device_class_hid_descriptor_send.c](https://github.com/azure-rtos/usbx/commits/master/common/usbx_device_classes/src/ux_device_class_hid_descriptor_send.c))

HaoboGu, 2023-04-06:
fixed by updating to latest version, thanks!

closed 2023-04-06 by HaoboGu