eclipse-threadx/usbx #98: USB host composite device

MihailZam, 2023-05-10:
Is it possible with the help of this stack to implement USB host composite device?

I have a SIM7600 USB modem. It is a composite device and have many CDC-ACM interfaces his bDeviceClass=0, bDeviceSubClass=0 and bDeviceProtocol=0 this mean - defined by the interface descriptors.
When device is connected stack calls ux_host_error_callback with eeror_code = UX_DEVICE_ENUMERATION_FAILURE because UX_NO_CLASS_MATCH(but CDC_ACM class registered).
I understand USB host composite device not implemented?

xiaocq2001, 2023-05-12:
What's the bInterfaceClass, bInterfaceSubClass and bInterfaceProtocol there? CDC-ACM in UX stack checks the class/subclass/protocol code there to see if it can be supported.

MihailZam, 2023-05-12:
`   ---------------------- Device Descriptor ----------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.0)
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0x1E0E (SHANGHAI BASECOM LTD.)
idProduct                : 0x9006
bcdDevice                : 0x0318
iManufacturer            : 0x01 (String Descriptor 1)
 Language 0x0409         : "SimTech, Incorporated"
iProduct                 : 0x02 (String Descriptor 2)
 Language 0x0409         : "SimTech, Incorporated"
iSerialNumber            : 0x03 (String Descriptor 3)
 Language 0x0409         : "0123456789ABCDEF"
bNumConfigurations       : 0x01 (1 Configuration)
------------------ Configuration Descriptor -------------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x02 (Configuration Descriptor)
wTotalLength             : 0x00B3 (179 bytes)
bNumInterfaces           : 0x04 (4 Interfaces)
bConfigurationValue      : 0x01 (Configuration 1)
iConfiguration           : 0x00 (No String Descriptor)
bmAttributes             : 0xA0
 D7: Reserved, set 1     : 0x01
 D6: Self Powered        : 0x00 (no)
 D5: Remote Wakeup       : 0x01 (yes)
 D4..0: Reserved, set 0  : 0x00
MaxPower                 : 0xFA (500 mA)
---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x00 (Interface 0)
bAlternateSetting        : 0x00
bNumEndpoints            : 0x02 (2 Endpoints)
bInterfaceClass          : 0xFF (Vendor Specific)
bInterfaceSubClass       : 0xFF
bInterfaceProtocol       : 0xFF
iInterface               : 0x00 (No String Descriptor)
---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x01 (Interface 1)
bAlternateSetting        : 0x00
bNumEndpoints            : 0x03 (3 Endpoints)
bInterfaceClass          : 0xFF (Vendor Specific)
bInterfaceSubClass       : 0x00
bInterfaceProtocol       : 0x00
iInterface               : 0x00 (No String Descriptor)
---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x02 (Interface 2)
bAlternateSetting        : 0x00
bNumEndpoints            : 0x03 (3 Endpoints)
bInterfaceClass          : 0xFF (Vendor Specific)
bInterfaceSubClass       : 0x00
bInterfaceProtocol       : 0x00
iInterface               : 0x00 (No String Descriptor)
---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x03 (Interface 3)
bAlternateSetting        : 0x00
bNumEndpoints            : 0x03 (3 Endpoints)
bInterfaceClass          : 0xFF (Vendor Specific)
bInterfaceSubClass       : 0x00
bInterfaceProtocol       : 0x00
iInterface               : 0x00 (No String Descriptor)
`

xiaocq2001, 2023-05-15:
As you can see all bInterfaceClass is 0xFF (Vendor Specific), which means the interface is vendor specific but not CDC-ACM, that's why interfaces are not recognized by existing classes.

xiaocq2001, 2023-07-17:
If there is no more comments, I'll close it in two days.

closed 2024-04-24 by rahmanih