eclipse-threadx/usbx #101: UsbX device printer

MarcoTrap, 2023-05-18:
I found a bug when I use printer class with other classes like one or more CDCs: the printer class must be not register as first to get the issue. 
When the usb device is connected to the host at the request GET_DEVICE_ID of the printer the device does not answer because can not find the class. 

I'm using the MIMXRT1170-EVK demoboard with MCUXpresso IDE v11.7.0 and SDK version 2.13.0 (667 2023-01-18).
I added a workaround to fix the problem in the file "ux_device_stack_control_request_process.c"
```c
                /* Is the request target to an interface?  */
                if ((request_type & UX_REQUEST_TARGET) == UX_REQUEST_TARGET_INTERFACE)
                {

                    /* Yes, so the request index contains the index of the interface 
                       the request is for. So if the current index does not match 
                       the request index, we should go to the next one.  */
                    /* For printer class (0x07) GET_DEVICE_ID (0x00) the high byte of 
                       wIndex is interface index (for recommended index sequence the interface
                       number is same as interface index inside configuration).  */
// WORKAROUND
                	// Check if it is a printer class (0x07) and if it is request GET_DEVICE_ID (0x00) then uses wIndex as big endian otherwise printer does not work with other devices
                 	if ((class_ptr -> ux_slave_class_interface -> ux_slave_interface_descriptor.bInterfaceClass == 0x07) && request == 0x00)                   
                        {
                            request_index = _ux_utility_short_get_big_endian(transfer_request -> ux_slave_transfer_request_setup + UX_SETUP_INDEX);
                         }
// WORKAROUND
                        if (((request_index & 0xFF) != class_index) ||
                            ((class_ptr -> ux_slave_class_interface -> ux_slave_interface_descriptor.bInterfaceClass == 0x07) &&
                             (request == 0x00) &&
                             *(transfer_request -> ux_slave_transfer_request_setup + UX_SETUP_INDEX + 1) != class_index))
                           continue;
                }
```
With this solution the usb printer device is acknowledge by host.

xiaocq2001, 2023-05-18:
Thanks for the feedback.
The printer case is actually checked by the code block after your workaround:
```
                        if (((request_index & 0xFF) != class_index) ||
                            ((class_ptr -> ux_slave_class_interface -> ux_slave_interface_descriptor.bInterfaceClass == 0x07) &&
                             (request == 0x00) &&
                             *(transfer_request -> ux_slave_transfer_request_setup + UX_SETUP_INDEX + 1) != class_index))
                           continue;

```
The mistake is "||", where should be "&&" to do not skip printer case.

xiaocq2001, 2023-05-19:
How about following check:
```
                    if (((request_index & 0xFF) != class_index) &&
                        !((class_ptr -> ux_slave_class_interface -> ux_slave_interface_descriptor.bInterfaceClass == 0x07) &&
                         (request == 0x00) &&
                         *(transfer_request -> ux_slave_transfer_request_setup + UX_SETUP_INDEX + 1) == class_index))
                        continue;
```
In case class index not match, the printer GET_DEVICE_ID case should be checked, if class index not match **AND** **NOT** printer GET_DEVICE_ID case, check next control.

xiaocq2001, 2023-05-19:
Sure, it will be fixed in next release.

closed 2023-05-19 by MarcoTrap

xiaocq2001, 2023-05-25:
I updated the code a bit for a better view, as following:
```
                    if ((request_type == 0xA1) && (request == 0x00) &&
                        (class_ptr -> ux_slave_class_interface -> ux_slave_interface_descriptor.bInterfaceClass == 0x07))
                    {

                        /* Check wIndex high byte.  */
                        if(*(transfer_request -> ux_slave_transfer_request_setup + UX_SETUP_INDEX + 1) != class_index)
                            continue;
                    }
                    else
                    {

                        /* Check wIndex low.  */
                        if ((request_index & 0xFF) != class_index)
                            continue;
                    }
```
In case request type and request is printer GET_DEVICE_ID the wIndex high byte is checked, otherwise wIndex low byte is checked as interface target.