eclipse-threadx/usbx #102: high speed usb cdc acm + msd disconnected after few minutes

calism, 2023-05-22:
I would to like to have usb cdc acm + msd on one usb port.

In the device manager, I see my USB serial device. My mass storage device also appears in the Windows Explorator. However after a few minutes, the USB serial device disappears from the Device Manager and likewise the mass storage from Windows Explorator. So it looks like usb gets disconnected after few minutes.

I use the following for the high speed device descriptor:

UCHAR device_framework_high_speed_cdc_acm[] = {
//========================
//Device Descriptor
0x12, // bLength
0x01, // bDescriptorType
D0(USB_HS_BCD_VERSION), // bcdUSB lo
D1(USB_HS_BCD_VERSION), // bcdUSB hi
0xEF, // DeviceClass
0x02, // DeviceSubClass
0x01, // DeviceProtocol
USB_HS_CONTROL_MAX_PACKET_SIZE, // bMaxPacketSize
D0(USB_DEVICE_VID), // idVendor lo
D1(USB_DEVICE_VID), // idVendor hi
D0(USB_DEVICE_PID), // idProduct lo
D1(USB_DEVICE_PID), // idProduct hi
0x00, // bcdDevice lo
0x01, // bcdDevice hi
0x01, // iManufacturer
0x02, // iProduct
0x03, // iSerialNumber
0x01, // bNumConfigurations
//==============================
// Device Qualifier Descriptor
0x0A, // bLength
0x06, // bDescriptorType
D0(USB_HS_BCD_VERSION), // bcdUSB lo
D1(USB_HS_BCD_VERSION), // bcdUSB hi
0x00, // DeviceClass
0x00, // DeviceSubClass
0x00, // DeviceProtocol
0x40, // bMaxPacketSize 8
0x01, // bNumConfigurations
0x00, // bReserved
//==============================
// Configuration Descriptor
0x09, // bLength
0x02, // bDescriptorType
// TO DO - Mustafa - this parameter needs to be modified when adding additional device
0x62, // wTotalLength lo
0x00, // wTotalLength hi
// TO DO - Mustafa - WITH MSD should be 3
0x03, // bNumInterfaces
0x01, // bConfigurationValue
0x00, // iConfiguration
0x40, // bmAttributes
0x32, // bMaxPower 0x32 * 2ma
//===== usb msd descriptor =====
// Interface descriptor
0x09,
0x04,
0x00,
0x00,
0x02,
0x08,
0x06,
0x50,
0x00,
// Bulk IN Endpoint descriptor
0x07, // bLength
0x05, // bDescriptorType
0x81, // bEndpointAddress
0x02, // bmAttribute
0x40, // wMAXPacketSize_lo
0x00, // wMAXPacketSize_hi
0x00, // bInterval
// Bulk OUT Endpoint descriptor
0x07, // bLength
0x05, // bDescriptorType
0x02, // bEndpointAddress
0x02, // bmAttribute
0x40, // wMAXPacketSize_lo
0x00, // wMAXPacketSize_hi
0x00, // bInterval
//===== ACM_1 descriptor =====
// Interface Association Descriptor (IAD)
0x08, // bLength
0x0B, // bDescriptorType
0x01, // bFirstInterface
0x02, // bInterfaceCount
0x02, // bFunctionClass
0x02, // bFunctionSubClass
0x00, // bFunctionProtocol
0x00, // iFunction
// Interface Descriptor
0x09, // bLength
0x04, // bDescriptorType
0x01, // bInterfaceNumber
0x00, // bAlternateSetting
0x01, // bNumEndpoints
0x02, // bInterfaceClass - CDC
0x02, // bInterfaceSubClass - Abstract Control Model
0x01, // bInterfaceProtocol
0x00, // iInterface
// CDC Class-Specific descriptor
// Header Functional Descriptor
0x05, // bLength
0x24, // bDescriptorType
0x00, // bDescriptorSubtype
0x10, // bcdCDC_lo
0x01, // bcdCDC_hi - specification version 1.10
// ACM Functional Descriptor
0x04, // bLength
0x24, // bDescriptorType
0x02, // bDescriptorSubtype
0x0e, // bmCapabilities
// Union Functional Descriptor
0x05, // bLength
0x24, // bDescriptorType
0x06, // bDescriptorSubtype
0x02, // bMasterInterface
0x01, // bSlaveInterface0
// Call Management Functional Descriptor
0x05, // bLength
0x24, // bDescriptorType
0x01, // bDescriptorSubtype
0x00, // bmCapabilities
0x01, // bDataInterface
// Notification Endpoint descriptor
0x07, // bLength
0x05, // bDescriptorType
0x84, // bEndpointAddress
0x03, // bmAttribute
0x08, // wMAXPacketSize_lo
0x00, // wMAXPacketSize_hi
0x08, // bInterval
// Data Class Interface Descriptor
0x09, // bLength
0x04, // bDescriptor
0x02, // bInterfaceNumber
0x00, // bAlternateSetting
0x02, // bNumEndpoints
0x0A, // bInterfaceClass
0x00, // bInterfaceSubClass
0x00, // bInterfaceProtocol
0x00, // iInterface
// Bulk IN Endpoint descriptor
0x07, // bLength
0x05, // bDescriptorType
0x85, // bEndpointAddress
0x02, // bmAttribute
D0(USB_HS_CDC_ACM_BULK_PACKET_SIZE), // wMAXPacketSize_lo
D1(USB_HS_CDC_ACM_BULK_PACKET_SIZE), // wMAXPacketSize_hi
0x00, // bInterval
// Bulk OUT Endpoint descriptor
0x07, // bLength
0x05, // bDescriptorType
0x04, // bEndpointAddress
0x02, // bmAttribute
D0(USB_HS_CDC_ACM_BULK_PACKET_SIZE), // wMAXPacketSize_lo
D1(USB_HS_CDC_ACM_BULK_PACKET_SIZE), // wMAXPacketSize_hi
0x00, // bInterval
};

Do you see an error somewhere? Any advice is welcome.

Best regards

xiaocq2001, 2023-05-23:
Since it's enumerated, the framework descriptors should be OK.
Do you have USB trace to check what happens when they disappear?
Will re-attaching make your device back (assuming board still power on when detaching USB)?

calism, 2023-05-23:
No I don't have a usb trace.
- If I unplug the USB cable and plug in again, the USB Serial Device does not appear in the Device Manager.
- A reset needs to be performed to have the USB Serial Device in the Device Manager again.

calism, 2023-05-24:
I have found an issue in my code and now everything is okay.

closed 2023-05-24 by calism