eclipse-threadx/usbx #105: USBX host hid mouse cause memory fault when using Logitech PRO X SUPERLIGHT mouse

xuzihan351, 2023-06-03:
Here is part of the Logitech PRO X SUPERLIGHT report descriptor:
[image]

Note the **Vendor-Defined 1** Usage Page has **Report Count(5)** and INPUT abs **Ary** data. But USBX alloc only **ONE** item for that. That makes memory out of bounds and thus lead to memory fault occur.
When deal with the report descriptors, USBX treat that as one item becase it don't have a **Usage Maximum**.
USBX use **Usage Maximum** to calc the ary item count, I don't know whether this is normative. Why not use Report Size and Report Count?

xiaocq2001, 2023-06-06:
Thanks for the feedback, we will check that.

xiaocq2001, 2023-07-03:
Simulated with the HID descriptor, enumeration seems good, can you give more details on how to reproduce the issue?

xuzihan351, 2023-07-03:
There is indeed no problem with the enumeration process here. The problem will occur in the message parsing the interrupt endpoint.I will send the message of the interrupt endpoint later.

xuzihan351, 2023-07-04:
The client buffer addr is 0x41c06e50 and its size is 0xA8, so its range is **_[0x41c06e50, 0x41C06EF8]_**. But when decompress, usbx try to write **_0x41c06f00_** which is a memory out of bounds.
Here is the log:
_ux_host_class_hid_transfer_request_completed 204 alloc client buffer addr(`0x41c06e50`) size(`0xA8`)
_ux_host_class_hid_report_decompress `raw data`: `00 00 00 00 00 00 00 00 01 93 40 00 00` 
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90001) to 0x41c06e50, filed value(0x0) to 0x41c06e54
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90002) to 0x41c06e58, filed value(0x0) to 0x41c06e5c
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90003) to 0x41c06e60, filed value(0x0) to 0x41c06e64
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90004) to 0x41c06e68, filed value(0x0) to 0x41c06e6c
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90005) to 0x41c06e70, filed value(0x0) to 0x41c06e74
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90006) to 0x41c06e78, filed value(0x0) to 0x41c06e7c
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90007) to 0x41c06e80, filed value(0x0) to 0x41c06e84
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90008) to 0x41c06e88, filed value(0x0) to 0x41c06e8c
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90009) to 0x41c06e90, filed value(0x0) to 0x41c06e94
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x9000a) to 0x41c06e98, filed value(0x0) to 0x41c06e9c
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x9000b) to 0x41c06ea0, filed value(0x0) to 0x41c06ea4
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x9000c) to 0x41c06ea8, filed value(0x0) to 0x41c06eac
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x9000d) to 0x41c06eb0, filed value(0x0) to 0x41c06eb4
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x9000e) to 0x41c06eb8, filed value(0x0) to 0x41c06ebc
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x9000f) to 0x41c06ec0, filed value(0x0) to 0x41c06ec4
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x90010) to 0x41c06ec8, filed value(0x0) to 0x41c06ecc
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x10030) to 0x41c06ed0, filed value(0x0) to 0x41c06ed4
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x10031) to 0x41c06ed8, filed value(0x0) to 0x41c06edc
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0x10038) to 0x41c06ee0, filed value(0x0) to 0x41c06ee4
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0xc0238) to 0x41c06ee8, filed value(0x0) to 0x41c06eec
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0xff000001) to 0x41c06ef0, filed value(0x1) to 0x41c06ef4
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0xff000093) to 0x41c06ef8, filed value(0x93) to **_0x41c06efc_**
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0xff000040) to **_0x41c06f00_**, filed value(0x40) to 0x41c06f04
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0xff000000) to **_0x41c06f08_**, filed value(0x0) to 0x41c06f0c
_ux_host_class_hid_field_decompress 148 decompress hid data, write field usage(0xff000000) to **_0x41c06f10_**, filed value(0x0) to 0x41c06f14

xiaocq2001, 2023-07-05:
Thanks for the logs. Please try to change code here:
https://github.com/azure-rtos/usbx/blob/7c928b43db68b72970b3effd5a2582eb5a6869c7/common/usbx_host_classes/src/ux_host_class_hid_report_add.c#L199, the number of items to decompress is not related to item type.
```
    /* Get the number of fields for this report.  */
    hid_field_count =  hid_parser -> ux_host_class_hid_parser_global.ux_host_class_hid_global_item_report_count;
```

xuzihan351, 2023-07-05:
This works. Thank you.

xiaocq2001, 2024-04-19:
Fix already in latest code. Can close.

closed 2024-04-24 by rahmanih