eclipse-threadx/usbx #109: Host stack HID support questions

sw2mrc, 2023-07-14:
Greetings!

I'm using stm32f7, and have few questions about using usbx host stack with HID devices,
and would appreciate your response.

1. When a composite HID device with two or more HID interfaces (say kbd and mouse) connects
   to stm32f7 host, only first interface gets recognized and configured, not the
   second interface. What api sequence needs to be used to enumerate all the interfaces
   and configure them?

2. Is there usbx host side api to retrieve all the descriptors that host stack queried and
   received, such as device, configuration, and HID report descriptors?

3. I'm able to receive events such as key press/release, mouse position etc, however,
   is there hostside api to retrieve the HID report sent by the device "as-is"?

Thanks in advance!

xiaocq2001, 2023-07-17:
1. Each HID interface should be enumerated as a HID class instance, you can get notified on HID instance creation in registered host change callback. Another way is to get the registered class (ux_host_stack_class_get) and get instances from the class (ux_host_stack_class_instance_get).
2. The descriptors are retrieved  for enumeration, there is no public to read them, but there is an API to read string descriptor, as ux_host_stack_device_string_get.
3. Register the HID class, do not register client but register report callback (ux_host_class_hid_report_callback_register), and start periodic report polling (ux_host_class_hid_periodic_report_start), you can handle report in callback.

sw2mrc, 2023-07-19:
Regarding 2: since there is no public api, are there any structures I can look at to come up with a private api if need be?

Regarding the other two items, I will test them in couple weeks and update the thread.

cheers.

sw2mrc, 2023-08-10:
>     3. Register the HID class, do not register client but register report callback (ux_host_class_hid_report_callback_register), and start periodic report polling (ux_host_class_hid_periodic_report_start), you can handle report in callback.

I was able to get this working with HID mouse where the report callback gets triggered, however it looks like the callback keeps getting triggered even when mouse is stationary and is not being used.. If the mouse is not being used, then there are only NAKs from the device, so I wonder why the report callback is being called continuously? Sometimes the reports seem to die down but I'm unable to see a pattern of when it would stop or runs continuously.

Overall, I'm looking for your guidance on which approach results in less MCU overhead when number of HID devices increase to say 4-7 devices? 1) a HID client thread that keeps polling for the mouse pointer/button/scroll wheel position continuously or 2) HID report callback invocation?

Thank you in advance..

xiaocq2001, 2023-08-11:
> however it looks like the callback keeps getting triggered even when mouse is stationary and is not being used.

If you have a USB tracer, please check to confirm that your mouse stops sending reports, if it sends reports in specific rate maybe you can set infinity idle rate with `ux_host_class_hid_idle_set()` ([ref here](https://github.com/azure-rtos/usbx/blob/7c928b43db68b72970b3effd5a2582eb5a6869c7/common/usbx_host_classes/src/ux_host_class_hid_mouse_activate.c#L153C1-L168C1)), if interrupt happens even there is no data, maybe the host controller driver (HCD) and STM32 HAL need to be checked.

sw2mrc, 2023-08-15:
> 1. Each HID interface should be enumerated as a HID class instance, you can get notified on HID instance creation in registered host change callback. Another way is to get the registered class (ux_host_stack_class_get) and get instances from the class (ux_host_stack_class_instance_get).

I've not been able to add support for a composite HID device with stm32/usbx host. Here are some details

1. Mouse-only HID device and Keyboard-only HID device works fine under usbx host
2. Composite Keyboard and Touchpad device (Rii x1) works fine with PC host, but under stm32/usbx host, only keyboard works, and touchpad doesn't work at all -- bottons, cursor pos, etc have no effect
3. In host_event_callback, there is usual sequence of events (HID_insert, USB_insert, and Device_connect) for both keyboard and mouse interfaces
4. Both keyboard and mouse are registered as HID clients and each have a separate user threads to poll keyboard and mouse status reports, and the polling happens only after device configure state is set in host_event_callback
5. Since keyboard works, i tried just forcing the kbd thread to sleep to see if mouse only actions can help retrieve the mouse state, but it doesn't work either.
6. Added a mutex to serialize simultaneous access in keyboard and mouse threads but it didn't help
7. Code does not crash

Please advise. Thank you in advance!

xiaocq2001, 2023-08-16:
> Composite Keyboard and Touchpad device (Rii x1) works fine with PC host, but under stm32/usbx host, only keyboard works, and touchpad doesn't work at all -- bottons, cursor pos, etc have no effect

Maybe keyboard and touchpad use different interfaces, the usage of touchpad does not match any existing registered client page and usage, so the touchpad is not recognized as a HID client.

> Both keyboard and mouse are registered as HID clients and each have a separate user threads to poll keyboard and mouse status reports, and the polling happens only after device configure state is set in host_event_callback
Since keyboard works, i tried just forcing the kbd thread to sleep to see if mouse only actions can help retrieve the mouse state, but it doesn't work either.
Added a mutex to serialize simultaneous access in keyboard and mouse threads but it didn't help

Not sure if the "mouse" here is real mouse or the touchpad, a touchpad may report itself a touchpad instead of a mouse, in this case there is no client driver matching and client instance is not available.

If it's confirmed the connected device is composed of HID keyboard interface and HID mouse interface, the HID class is registered, and HID mouse and keyboard clients are registered, the keyboard and mouse client instances should be recognized.

sw2mrc, 2023-08-16:
> Maybe keyboard and touchpad use different interfaces, the usage of touchpad does not match any existing registered client page and usage, so the touchpad is not recognized as a HID client. May you share the descriptors for the composite device to check?

Attached is the descriptors from usbview.

> If it's confirmed the connected device is composed of HID keyboard interface and HID mouse interface, the HID class is registered, and HID mouse and keyboard clients are registered, the keyboard and mouse client instances should be recognized.

it would be sufficient if usbx supported boot interface.. I was unable to find if usbx supports it or not..
[rii-x1-combo-kbd-mse.txt](https://github.com/azure-rtos/usbx/files/12357456/rii-x1-combo-kbd-mse.txt)

sw2mrc, 2023-08-29:
@xiaocq2001 any updates on the response i posted couple weeks ago? thank you.

xiaocq2001, 2023-08-30:
Sorry for late reply.

There are configuration descriptors in attached file, but do you have HID report descriptor? The contents in report descriptor are used to recognize clients and define report layouts.

For "boot interface" do you mean boot protocol devices? Unfortunately boot protocol is not enabled by default, and there is no API to switch to boot protocol for now.

sw2mrc, 2023-08-30:
HID report descriptor captured from wireshark attached.

> For "boot interface" do you mean boot protocol devices? Unfortunately boot protocol is not enabled by default, and there is no API to switch to boot protocol for now.

Is there a plan for it to be enabled? it would be a very useful feature if usbx supported it..

[hid-report-desc-rii-combo-kbd-mse.txt](https://github.com/azure-rtos/usbx/files/12478097/hid-report-desc-rii-combo-kbd-mse.txt)

xiaocq2001, 2023-09-01:
I checked the stack by simulate composite device, there seems no problem on the two interfaces operations. Maybe you can check if there is issue on STM HAL or DCD part on enabling two interrupt pipes.

sw2mrc, 2023-09-01:
> I checked the stack by simulate composite device, there seems no problem on the two interfaces operations. Maybe you can check if there is issue on STM HAL or DCD part on enabling two interrupt pipes.

Thank you for the confirmation.

1. Could you please share the test stub so i can see if there are any issues from my side before I reach out to ST?
2. If sharing code is not possible, could you confirm if both input reports from mouse and keyboard need to be processed from a single thread or can they be in two separate threads? My code processes input reports from two separate threads and since it is just one device instance but two interfaces, i was wondering if there is a data race issue elsewhere as two threads in my code use mutex

Regards..

xiaocq2001, 2023-09-04:
Testing code is as following, but I'm afraid you can not use that directly, since it requires simulator implement.

```
/* This test is designed to test simple usage of the composite (mouse + keyboard) class.  */

#include "ux_api.h"
#include "ux_device_class_hid.h"
#include "ux_host_class_hid.h"

#include "ux_host_class_hid_keyboard.h"
#include "ux_host_class_hid_mouse.h"

#include "ux_test_utility_sim.h"
#include "ux_test_hcd_sim_host.h"

#define     UX_DEMO_STACK_SIZE  1024
#define     UX_DEMO_MEMORY_SIZE (96*1024)

#define     ARRAY_COUNT(array)  (sizeof(array)/sizeof(array[0]))
#define     LSB(x)              (x & 0xff)
#define     MSB(x)              ((x & 0xff00) >> 8)

static void tx_demo_thread_host_simulation_entry(ULONG);
static void tx_demo_thread_device_simulation_entry(ULONG);

static UINT demo_thread_hid_keyboard_callback(UX_SLAVE_CLASS_HID *, UX_SLAVE_CLASS_HID_EVENT *);
static UINT demo_thread_hid_mouse_callback(UX_SLAVE_CLASS_HID *, UX_SLAVE_CLASS_HID_EVENT *);

static UCHAR                        usbx_memory[UX_DEMO_MEMORY_SIZE + (UX_DEMO_STACK_SIZE * 2)];
static TX_THREAD                    tx_demo_thread_host_simulation;
static TX_THREAD                    tx_demo_thread_device_simulation;

static UX_HOST_CLASS_HID_KEYBOARD   *hid_keyboard;
static UX_HOST_CLASS_HID_MOUSE      *hid_mouse;

static UX_SLAVE_CLASS_HID_PARAMETER hid_parameter;

TX_EVENT_FLAGS_GROUP                ev_flags;
#define EV_KEYBOARD_RECEIVED        0x01u
#define EV_MOUSE_RECEIVED           0x02u

/* REF: Composite Keyboard and Touchpad device (Rii x1)  */

static UCHAR hid_keyboard_report[] = {
    0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
    0x09, 0x06,        // Usage (Keyboard)
    0xA1, 0x01,        // Collection (Application)
    0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
    0x19, 0xE0,        //   Usage Minimum (0xE0)
    0x29, 0xE7,        //   Usage Maximum (0xE7)
    0x15, 0x00,        //   Logical Minimum (0)
    0x25, 0x01,        //   Logical Maximum (1)
    0x75, 0x01,        //   Report Size (1)
    0x95, 0x08,        //   Report Count (8)
    0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x95, 0x01,        //   Report Count (1)
    0x75, 0x08,        //   Report Size (8)
    0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x95, 0x05,        //   Report Count (5)
    0x75, 0x01,        //   Report Size (1)
    0x05, 0x08,        //   Usage Page (LEDs)
    0x19, 0x01,        //   Usage Minimum (Num Lock)
    0x29, 0x05,        //   Usage Maximum (Kana)
    0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0x95, 0x01,        //   Report Count (1)
    0x75, 0x03,        //   Report Size (3)
    0x91, 0x03,        //   Output (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0x95, 0x06,        //   Report Count (6)
    0x75, 0x08,        //   Report Size (8)
    0x15, 0x00,        //   Logical Minimum (0)
[... 605 lines cut ...]
                ux_utility_memory_set(hid_event.ux_device_class_hid_event_buffer, 0, 8);
                /* Report ID.  */
                hid_event.ux_device_class_hid_event_buffer[0] = 1;
                /* Change the buttons.  */
                hid_event.ux_device_class_hid_event_buffer[1] = mouse_buttons ++;
                /* Change the x, y, and wheel positions. These are relative values.  */
                hid_event.ux_device_class_hid_event_buffer[2] = 1;
                hid_event.ux_device_class_hid_event_buffer[3] = 1;
                hid_event.ux_device_class_hid_event_buffer[4] = 1;

                /* Set the mouse event.  */
                ux_device_class_hid_event_set(hid, &hid_event);
            }

            /* Wait for host to receive.  */
            tx_event_flags_get(&ev_flags, EV_MOUSE_RECEIVED | EV_KEYBOARD_RECEIVED,
                                        TX_OR_CLEAR, &flags, TX_WAIT_FOREVER);

        }
    }
}

```

The APIs to get input reports from mouse and keyboard are actually not blocking, so they can be processed in single thread (in my test code `ux_host_class_hid_keyboard_key_get` is used to check if there is key ready in key queue, `ux_host_class_hid_mouse_buttons_get`, `ux_host_class_hid_mouse_position_get` and `ux_host_class_hid_mouse_wheel_get` are used to get current state for the mouse).

sw2mrc, 2023-09-06:
@xiaocq2001 thanks for the details. I will update this thread (or close the issue) in a week. cheers.

closed 2023-09-14 by sw2mrc