eclipse-threadx/usbx #113: ux_dcd_... files missing in usbx 6.2.1 ?

roblxx, 2023-09-01:
Hi,

In older versions of usbx (e.g., 5.7) many device controller files (ux_dcd_mcf5329_... files for example) were provided with the usbx sourc code. In usbx 6.2.1 those files are not present - there appears to be only a sample ux_dcd set of files.

Can you tell me if the previously provided ux_dcd ... device controller files are available for usbx 6.2.1, or if not, can you point me to documentation that indicates where to find them or how to create them?

Thanks,
Rob

xiaocq2001, 2023-09-07:
Thanks for the feedback.

Unfortunately, the DCD device controller drivers are not available for USBX 6.x. Usually, they are packed by semi vendor that provides USBX in their packages (E.g., in STM32 Cube, NXP software packet, Reneses SSP etc.).

You can still use older version DCD files and update them to fit USBX 6.x.

roblxx, 2023-09-07:
Thanks for the feedback – kind of what we thought but good to get it confirmed.

Rob

[ EXTERNAL ]