eclipse-threadx/usbx #118: Getting one issue in USB host uvc support

Mahesh-dev-avula, 2023-09-29:
@xiaocq2001 ,

I am working on USB host uvc support, usb devices as connecting to my target board are logitch camera and AIRHUG camera ,
1. With logitech camera it's working fine with resolution as follows below: 680*480 with 30fps and 1280*720 with 10fps,
2. but, if I use resolution as 1280*720 with 30fpsfps since usb camera device supports, it's not working and observed that there is no valid frames.
3. I could see that it's throwing **RING OVERRUN_EVENT**.
4. From the USB trace i have observed that when multi transactions per microframe scenorio it's not working, but its' working only when single transaction per microframe.
5. I have gone through the usbx host video class code and understand that at time only one payload is requesting to the usb camera device, this should be fine only when single tracsactions per microframe but, I believe that when we increase the resolution and framerate, we have to request mutilple payloads at time.

Please let me know if my understanding is correct, also could you please suggest that how we can request mutilple payloads at time from the application.

Regards
Mahesh

Mahesh-dev-avula, 2023-10-05:
Hi @xiaocq2001,@yuxin-azrtos,

Any update on above comment.

xiaocq2001, 2023-10-11:
`ux_host_class_video_transfer_buffers_add` adds a list of buffers before transfer start, each buffer is for a video payload including 1 to 3 transactions. When a buffer is sent callback set by `ux_host_class_video_transfer_callback_set` is called, to append a new buffer to end of the list. In this way certain number of buffers are "cached" for sending.

Multiple transactions per microframe is usually supported by controller driver, to allow controller endpoint to filled in multiple banks, or FIFO filled by 1~3 transactions bytes, or filled by DMA by 1~3 transaction bytes, if the controller driver is not modified to fit the multiple transactions transfer support that is not working.

Maybe you can check if your hardware (controller) can support multiple transactions per microframe and if your controller driver implements code to support this function.

Mahesh-dev-avula, 2023-10-11:
> `ux_host_class_video_transfer_buffers_add` adds a list of buffers before transfer start, each buffer is for a video payload including 1 to 3 transactions. When a buffer is sent callback set by [...]

Yes, it's correct, you mean that same application will work for single transaction per microframe and multi transactions(1 - 3) per also, we no need to prepare multiple buffers at time for multi transactions per microframe.
I really appriciate your help on this.
Regards
Mahesh