eclipse-threadx/usbx #121: standalone usbx host stack support

mendez23, 2023-10-21:
Greetings!

Is USBX host stack supported in standalone mode, i.e, without using Threadx or an RTOS?

Thank you!

BluesharkPD, 2023-10-21:
Yes it work's for host class hid, storage, cdc acm, printer, hub

mendez23, 2023-10-23:
@BluesharkPD , thanks for the confirmation.

I have few follow-on questions:

1. Is hub class supported as part standalone usbx host stack?
2. Is composite hid device as part of device stack supported?
2. Since device stack can also be standalone, can both device and host stack be used together as standalone?
3. I'm assuming that using other rtos like freertos is still ok, but usbx stack itself is single threaded, correct?

I would really appreciate some clarity here, and thank you in advance!

BluesharkPD, 2023-10-26:
@mendez23 which mcu are you using??,  you need to enchure that usbx controllers support standalone mode

BluesharkPD, 2023-10-26:
1. Yess hub class is supported in standalone mode.
2. Most of USBX device class work in standalone.
3. I guess yess since otg mode is supported.
4. In Rtos mode only thread mx is supported but in standalone mode i suppose that if you put usb process in a thread it it will work fine.

mendez23, 2023-10-26:
@BluesharkPD thanks for the clarifications! currently using stm32f746 for prototyping -- usbx examples work, but have not tried standalone mode..

BluesharkPD, 2023-10-26:
As I khnow the last x-cube-azrtos-f7 is not up to date so I recommende to refer to h7 azrtos package it support standalone and update middleware and controllers interface. There's also standalone host and device application. For Stm32h7 and stm32F7 it's the same ip so application could be similar.

mendez23, 2023-11-01:
@BluesharkPD thanks. Will check them out.

closed 2023-11-01 by mendez23