eclipse-threadx/usbx #128: usbx host stack behavior

sw2mrc, 2023-11-24:
Hello,

I have few clarification questions on usbx host stack behavior..

Assuming there are two devices and a hub on usb 2.0 bus, I would like some clarity on the host stack behavior:

1. Will usbx allocate an EP0 channel for each device, so, in this case three ep0 channels?
2. Will these three ep0 channels remain in allocated state and not available for reassignment until these devices are detached from the bus?

A separate question -- it seems like the host stack processes periodic edlist even where there are no devices on the bus - shouldn't it skip the processing task altogether when there are no devices on the bus?

Thank you.

sw2mrc, 2023-12-11:
Hello, I would appreciate a response.. thank you.

xiaocq2001, 2023-12-12:
The action actually depends on the implement in HCD,
1. Yes, the host stack allocates EP0 for each device and issues command to HCD to create physical channel for each EP0
2. HCD allocates physical channel for each EP0 (the current implement) but it's possible for HCD to maintain logical EP0 to share physical channels for all EP0s on all devices.

closed 2023-12-17 by sw2mrc