eclipse-threadx/usbx #130: AOAP Feature

ch871424956, 2023-12-13:
1. STM32H573 chip is used

2. Integration of threadx and usbx

3. usb aoa function is required to communicate with Android phones

Do you have usbx with aoap?
Do you have a usbx development template if you write it yourself?

Thanks!

xiaocq2001, 2023-12-14:
Unfortunately, AOA is not supported directly right now, if the connected Android phone device has standard HID/Audio/Video interfaces they can be recognized if corresponding class is registered.
To develop new class by yourself, you can refer to existing class implements:
* You can check `ux_host_class_asix` for method to recognize the device by its VID PID;
* You can check `ux_host_class_cdc_acm` for method to recognize the device by its interface class/sub class/protocol codes.

111llll111, 2023-12-18:
Hello CQ，

I refer to USBX to write AOA Class, and get two endpoints through _ux_host_stack_interface_endpoint_get. Data will be sent and received through these two endpoints. Is there a way to send data using endpoint 0? We don't use these endpoints for sending and receiving data.

Thanks！

xiaocq2001, 2023-12-18:
You can refer to https://github.com/azure-rtos/usbx/blob/f856e41e7fbc4620e12faba04c97c2da6abc10fc/common/usbx_host_classes/src/ux_host_class_hid_idle_set.c#L119C1-L146C65, for each device, the EP0 instance is in `UX_DEVICE::ux_device_control_endpoint`, since EP0 is shared by all interfaces, `UX_DEVICE::ux_device_protection_semaphore` is used to lock it to avoid access confliction.

111llll111, 2023-12-19:
Hello CQ，

I sent AOA protocol configuration through EP0, and started AOA, the phone successfully switched into Android accessory mode, but I referred to **__ux_host_class_cdc_acm_write_** function to send data to the phone, logical analysis to grab address=0x01, endpoint=0x02, PID DATA0 has data, but the phone does not receive data. What is the reason?

Thanks！

111llll111, 2023-12-20:
Hello CQ，

I see PID=0x2d01, which is ADB+ACCESSORY, so there should be two interfaces. How to get these two interfaces and distinguish them in USBX?

Thanks！

111llll111, 2023-12-20:
Hello CQ,

I have successfully debugged AOA based on USBX. Thank you very much for your support!

closed 2024-01-11 by 111llll111