eclipse-threadx/usbx #133: USB host MSC class READ FORMAT CAPACITY command failing.

Mahesh-dev-avula, 2024-01-11:
Hi, I am working on USB host **mass storage class**, as part class request there was a command **READ FORMAT CAPACITY** which is sending from host to device, but device endpoint getting stalled and at the same time host side also STALL ENDPOINT event gets generated and it's trying to reset the endpoint.

below is the code where we can send the READ FORMAT CAPACITY command to the device.

  **/* Some devices require we do this.  */
        status =  _ux_host_class_storage_media_format_capacity_get(storage);
        if (status != UX_SUCCESS)
            return(status);**
One thing I could say that if we comment above snippet code, MSC class working fine.
Could you please help on this if we can comment this code and continue further MSC class.

->I am using usbx version is v6.1.9_rel.
->ARMCLANG 
I am attaching USB trace screenshot for your reference.
[image]

Regards
Mahesh

Mahesh-dev-avula, 2024-01-11:
Hi @@xiaocq2001 ,
Could you please help on above issue.

xiaocq2001, 2024-01-11:
I'm not sure if it's required for some old mass storage device to enumerate, maybe you can try to ignore the error and just continue the steps (dont check the error code to return).

Mahesh-dev-avula, 2024-01-11:
Hi @xiaocq2001,
Thank you so much for update.