eclipse-threadx/usbx #137: WCID support

gamelaster, 2024-01-16:
Hello, I would like to ask, what is the best way of implementing WCID (thus auto installing WinUSB driver) in USBX?
Currently, we have been able to get it working, but we needed to modify USBX source to workaround ascii to unicode conversion of strings, but we would like to avoid modifying usbx, so having proper way of doing this would be much better. Thanks!

xiaocq2001, 2024-01-17:
Currently USBX accepts ascii string and extend it to unicode string (padding zero on each ascii byte) defined by USB spec.
The WCID have a vendor ID and a padding byte after the unicode string (referencing https://github.com/pbatard/libwdi/wiki/WCID-Devices#user-content-Microsoft_OS_String_Descriptor),
so it's worthy to try the string framework like this:
```c
UCHAR string_framework[] = { 

    /* Manufacturer string descriptor : Index 1 */
        0x09, 0x04, 0x01, 0x0c, 
        0x45, 0x78, 0x70, 0x72,0x65, 0x73, 0x20, 0x4c, 
        0x6f, 0x67, 0x69, 0x63,

    /* Product string descriptor : Index 2 */
        0x09, 0x04, 0x02, 0x0c, 
        0x44, 0x61, 0x74, 0x61, 0x50, 0x75, 0x6d, 0x70, 
        0x44, 0x65, 0x6d, 0x6f,  

    /* Serial Number string descriptor : Index 3 */
        0x09, 0x04, 0x03, 0x04, 
        0x30, 0x30, 0x30, 0x31
    };

    /* WCID MS OS String Descriptor.  */
        0x09, 0x04, 0xEE,
        0x08, /* 8 bytes to extend (including vendor code byte)  */
        'M', 'S', 'F', 'T', '1', '0', '0', '0',
        VendorCode /* Vendor code */

```

majsi, 2024-01-18:
Thank you for your comment. If I modify the code from your answer like this:
`
UCHAR string_framework[] = { 
    /* Manufacturer string descriptor : Index 1 */
        0x09, 0x04, 0x01, 0x0c, 
        0x45, 0x78, 0x70, 0x72,0x65, 0x73, 0x20, 0x4c, 
        0x6f, 0x67, 0x69, 0x63,

    /* Product string descriptor : Index 2 */
        0x09, 0x04, 0x02, 0x0c, 
        0x44, 0x61, 0x74, 0x61, 0x50, 0x75, 0x6d, 0x70, 
        0x44, 0x65, 0x6d, 0x6f,  

    /* Serial Number string descriptor : Index 3 */
        0x09, 0x04, 0x03, 0x04, 
        0x30, 0x30, 0x30, 0x31,

    /* WCID MS OS String Descriptor.  */
        0x00, 0x00, 0xEE,
        0x08, /* 8 bytes to extend (including vendor code byte) */
        'M', 'S', 'F', 'T', '1', '0', '0',
        <VendorCode> /* Put your vendor code here */

    };
`
then MS Windows correctly reads the 0xEE string descriptor. Software such as USB Device Tree Viewer and Thesycon USB Descriptor Dumper do not display this descriptor because they read 0xEE with a language ID other than 0x0000.

The macro you mentioned,
UX_DEVICE_ENABLE_GET_STRING_WITH_ZERO_LANGUAGE_ID,
I couldn't find in the firmware package I am using, available at [STM32CubeH5 v1.1.0](https://github.com/STMicroelectronics/STM32CubeH5/releases/tag/v1.1.0). When I compared the codes with eclipse-usbx, I noticed the following section is missing:

`
#ifdef UX_DEVICE_ENABLE_GET_STRING_WITH_ZERO_LANGUAGE_ID

            /* Check if the language ID is zero.  */
            if (request_index == 0)
            {

                /* Get the first language ID in the language ID framework.  */
                request_index =  _ux_utility_short_get(_ux_system_slave -> ux_system_slave_language_id_framework);
            }
#endif
`