eclipse-threadx/usbx #142: Support other RTOS

EmbWareHouse, 2024-02-01:
Hi, I have a question for threadx-eclipse teams
It is possible to rework architecture to support other RTOS ? (i see that USBX will be ported to PX5)  so it is possible to do that with other RTOS like POSIX or FreeRTOS or chibios ??

EmbWareHouse, 2024-10-02:
Any chance to support it. 
I know that it can break the actual stack api/architect but it target more embedded users.

fdesbiens, 2024-11-25:
Hi @EmbWareHouse.

For the time being, ThreadX and its companion modules are maintained by the same small team. We cannot currently provide support for other RTOSes. That said, I am personally not against it in principle. We would consider seriously code contributions that add explicit support for other platforms. On the other hand, such support would be out of the scope of our safety certification efforts.

fdesbiens, 2025-04-04:
Closing this issue due to a lack of activity. Feel free to reopen.

closed 2025-04-04 by fdesbiens