eclipse-threadx/usbx #172: How to use  the functions  ux_system_initialize  about  cache_safe_VOID *cache_safe_memory_pool_start, ULONG cache_safe_memory_size

andy817501, 2024-09-06:
i have open the dcahe and setted it, but i I don't know How to provide a pointer, and the Cache how to use in usbx

ayedm1, 2024-09-06:
Hi @andy817501, 
You can this example from NXP 

``
#ifndef USBX_MEMORY_SIZE
#define USBX_MEMORY_SIZE            (60 * 1024)
#endif

#ifndef USBX_MEMORY_CACHESAFE_SIZE
#define USBX_MEMORY_CACHESAFE_SIZE  (60 * 1024)
#endif

ULONG usb_memory[USBX_MEMORY_SIZE / sizeof(ULONG)];
AT_NONCACHEABLE_SECTION_ALIGN(char usb_memory_cachesafe[USBX_MEMORY_CACHESAFE_SIZE], 64);

    ux_system_initialize((VOID *)usb_memory, USBX_MEMORY_SIZE,
                         usb_memory_cachesafe, USBX_MEMORY_CACHESAFE_SIZE);`

``

andy817501, 2024-09-11:
Thank you for your help

andy817501, 2024-09-12:
> 你好@andy817501， 你可以从 NXP 获得这个例子
> 
> https://github.com/nxp-mcuxpresso/mcux-sdk-examples/blob/fc18ff0c12ba90fe71714eaa4244717efa18c61c/evkmimxrt1040/azure_rtos_examples/usbx_host_hid_mouse/board_setup.c#L89
> 
> ` #如果定义 USBX_MEMORY_SIZE #定义 USBX_MEMORY_SIZE (60 * 1024) #结束
> 
> #如果定义 USBX_MEMORY_CACHESAFE_SIZE #定义 USBX_MEMORY_CACHESAFE_SIZE (60 * 1024) #结束
> 
> ULONG usb_memory[USBX_MEMORY_SIZE / sizeof(ULONG)]; AT_NONCACHEABLE_SECTION_ALIGN(char usb_memory_cachesafe[USBX_MEMORY_CACHESAFE_SIZE], 64);
> 
> ```
> ux_system_initialize((VOID *)usb_memory, USBX_MEMORY_SIZE,
>                      usb_memory_cachesafe, USBX_MEMORY_CACHESAFE_SIZE);`
> ```
> 
> `

I would like to ask if there are any precautions to be taken with USBX if I want to use DMA

closed 2024-09-23 by andy817501