eclipse-threadx/usbx #188: Does USB_CDC_ACM host support hot-plugging?

andy817501, 2025-02-20:
I did not find any information about hot plugging in the document. During my testing, I found that after the host communicates with the device, disconnecting the device will prevent it from connecting properly

fdesbiens, 2025-04-04:
At a minimum, we could improve the documentation around this, @andy817501. I will discuss this with the team.

andy817501, 2025-04-10:
Thank you for your reply

fdesbiens, 2025-04-15:
Can you provide additional details per my previous comment, @andy817501?

andy817501, 2025-04-22:
I am using USB_CDC_ACM_HOST  , Using USB full speed interface, i am trying to plug USB_CDC_ACM_DEVICE and it is USB full speed interface too

ayedm1, 2025-04-22:
Hi @andy817501 which plateforme are you using ?

andy817501, 2025-04-27:
> 你好[@andy817501](https://github.com/andy817501)您使用的是哪个平台/目标？Stm32？

 yes i used stm32

andy817501, 2025-04-27:
> Hi [@andy817501](https://github.com/andy817501) which plateforme/target are you using ? Stm32 ?

I am using vscode + gcc + stm32

fdesbiens, 2026-09-28:
USBX has a CDC-ACM host deactivation path. To investigate why the device does not reconnect, could you share the exact STM32 part, USBX version, host-controller driver, and the host insertion/removal callback events or error statuses from the first disconnect and second connection?