eclipse-threadx/usbx #195: Potential bug in time difference macro _ux_utility_time_elapsed in USBX protocol stack

rsang0, 2025-04-16:
Describe the bug
Hello,

I have noticed a potential issue with the macro `_ux_utility_time_elapsed` which appears in the USBX protocol stack source code on GitHub. The macro is intended to calculate the elapsed time between two timestamp values, considering a 32-bit overflow.

The current implementation is:

#define _ux_utility_time_elapsed(a, b) \
    (((b) >= (a)) ? ((b) - (a)) : (0xFFFFFFFFul - (b) + (a) + 1))

Issue:

The calculation of the overflow case is incorrect because it swaps the roles of `a` and `b` in the overflow scenario. Specifically, when `b < a`, the elapsed time should be calculated as `0xFFFFFFFFul - a + b + 1`, assuming `a` is the earlier timestamp and `b` is the later timestamp that has caused an overflow.

However, the current macro computes it as `0xFFFFFFFFul - b + a + 1`, which effectively reverses the order of `a` and `b` in the calculation, leading to incorrect results in cases where `a` and `b` are not in the expected order.

Example:

Suppose `a = 0xFFFFFF00` (near overflow), and `b = 0x00000100` (after overflow).  
- Correct elapsed time should be:  
  `0xFFFFFFFFul - a + b + 1 = 0xFFFFFFFF - 0xFFFFFF00 + 0x100 + 1 = 255 + 256 + 1 = 512`  
- But with the current macro:  
  `0xFFFFFFFFul - b + a + 1 = 0xFFFFFFFF - 0x100 + 0xFFFFFF00 + 1`, which results in an incorrect value.

Potential Impact:

This logic error can cause the elapsed time calculation to be incorrect in scenarios involving overflow, leading to issues in timeouts, time measurements, or synchronization.

Suggestion:

The macro should be corrected to:

#define _ux_utility_time_elapsed(a, b) \
    ( ((b) >= (a)) ? ((b) - (a)) : (0xFFFFFFFFul - (a) + (b) + 1) )

which correctly calculates the elapsed time assuming `a` is the earlier timestamp and `b` the later one, considering a single overflow.

fdesbiens, 2025-04-22:
Hi @rsang0. Thank you for submitting this issue. We will discuss it during our weekly meeting and get back to you.

@eclipse-threadx/iot-threadx-committers and @eclipse-threadx/iot-threadx-contributors: please review.

fdesbiens, 2025-05-13:
This issue has been resolved by PR #198. Merged to the dev branch today. Will ship in the next service release.

PR #198 (2025-05-13): Fix/time elapsed
Fixes #195 with regression tests.
closed 2025-05-13 by fdesbiens