eclipse-threadx/usbx #206: Possible buffer overflow

adob, 2025-09-19:
`_ux_device_stack_initialize` allocates 256 bytes for `ux_slave_transfer_request_data_pointer` but up to 512 bytes may be written there since wMaxPacketSize is 512 in USB High-Speed mode.

Size check here:
https://github.com/eclipse-threadx/usbx/blob/cfdd55c48a2f44afa55cdb198fe65ddb0c8565ba/common/core/src/ux_device_stack_initialize.c#L186-L187

Buffer allocated here:
https://github.com/eclipse-threadx/usbx/blob/cfdd55c48a2f44afa55cdb198fe65ddb0c8565ba/common/usbx_device_classes/src/ux_device_class_cdc_acm_read.c#L190-L194

Buffer written here:
```
 HAL_PCD_EP_Receive(dcd_stm32 -> pcd_handle,
                            endpoint->ux_slave_endpoint_descriptor.bEndpointAddress,
                            transfer_request->ux_slave_transfer_request_data_pointer,
                            transfer_request->ux_slave_transfer_request_requested_length);
```

fdesbiens, 2025-10-14:
Thank you for raising this issue @adob. I will ask a member of the team to take a look.

fdesbiens, 2026-09-28:
I’m sorry for the slow follow up. I checked the buffer paths in current USBX. The 256 byte allocation is for the control endpoint; CDC ACM reads use a separate bulk OUT endpoint buffer. The shipped core owned buffers are at least 1024 bytes, and the class owned CDC ACM read buffer defaults to 512 bytes. A standard high speed CDC ACM packet of 512 bytes therefore fits in either default buffer.

I’m closing this for now. If you’re seeing an overflow with a custom configuration, please open another issue with the buffer settings, endpoint descriptor, and a reproducer so we can investigate that case.

closed 2026-09-28 by fdesbiens